elementary/camera #7: There's no feedback when taking a photo

danrabbit, 2016-10-02:
Photos are succesfully taken, but there's no feedback to indicate that they were. There should probably be some kind of audio cue (shutter sound) and visual cue ("flash" to black so that we don't actually flash and mess up the photo)

alainm23, 2018-05-27:
This could work with a set brightness to the maximum for 0.5 second and return to normal, it's an idea, I'll see how to do it

alainm23, 2018-05-27:
The same thing happens when a video is recorded, it is not known if it is saved, you could add a notification indicating the file path

robinkooli, 2018-09-13:
Show a thumbnail of the taken picture on left bottom?

pedropaulosuzuki, 2022-01-18:
https://github.com/elementary/camera/commit/32e4c613ab19a67e660faed4585aa4ff88ff128f

Did this ever become a thing? It seems to kinda fix this issue. (Did not tested it myself).

PR #223 (2022-01-24): Ensure pipeline restarts and controls function after taking photo
Fixes #221
Incidentally fixes #7 (?)

This recreates the current pipeline after take photo instead of setting the state to PLAYING (which does not seem to work after setting the state to NULL).

This incidentally provides a possible fix for #7 since the preview goes black while the pipeline is recreated.  The sound effect is played just before recreating the pipeline so the sound and visual effects are more closely synchronized.

I suspect more elegant solutions are possible but I am not sufficiently knowledgeable of GStreamer a [... cut]
closed 2022-01-24 by danrabbit