elementary/camera #35: Video is really choppy with Logitech C920

AndrewVos, 2017-06-28:
### Steps to reproduce the issue

1. Plug in a Logitech C920 

### Expected behavior

Video to not be really slow.

### System information

- **elementary OS 0.4.1 Loki**:

I also tried installing `guvcview` from package and from the ppa on  sourceforge, no difference. Also there are some mentions online of changing `Auto white balance` in `guvcview`, but this didn't change anything.

In `cheese` and in `Chromium` there is no problem. Audio and video works as expected.

davidak, 2017-08-23:
I have the same issue on all Notebooks internal webcams.

- MacBook Pro 10.2
- Dell Latitude E5470

Way better quality in Wire (messanger) and Chromium.

BrusAndr, 2017-09-08:
+1, Logitech C920, video is very slow
elementary OS 0.4.1 Loki stable

matzipan, 2017-09-09:
Welp, since you already have the hardware, it would be very useful if you could have a go at trying to fix the issue. 

If you need any help figuring out the issue, just ask.

BrusAndr, 2017-09-09:
Sorry, I'm not a developer yet.

artemanufrij, 2018-03-30:
@AndrewVos could you test #44 please?

AndrewVos, 2018-03-30:
Sorry @artemanufrij I don't run Elementary anymore

artemanufrij, 2018-03-30:
@davidak could you test it please?

davidak, 2018-03-30:
@artemanufrij i also don't run elementary OS and don't have the devices anymore.

Are you changing the resolution of the preview or also the final recording? The latter would be bad.

Schyken, 2018-11-03:
I'd like to request this issue be reopened. I have a Logitech C920 and experience this issue myself. I'd be willing to do additional testing and provide logs. The issue is similar to the original post with the only exception being that other applications like GUVCVIEW working as expected. (My guess based on what I've seen in applications like OBS, is that the app is pulling in the highest possible resolution for the camera to output, which is *not* good as that resolution runs at 2-5FPS max. Possibly having options to change the resolution in-app may help.)

cassidyjames, 2018-12-21:
@Schyken this issue was never closed.

While investigating this, I found that Cheese (and likely other apps) use a lower resolution by default which does not cause problems. In Cheese, choosing the camera's highest resolution produces similarly choppy and freezing results. So I think the resolution (heh) to this issue would be to cap out on a smaller resolution, perhaps checking what Cheese is doing.

Schyken, 2018-12-21:
I apologize @cassidyjames, I must have read that wrong (I still don't fully understand the way GitHub threads work)

Thanks quite a bit for the response, I will try and look deeper into it and hopefully be able to provide (hopefully) useful feedback.

totakaro, 2019-04-24:
Hi, new Elementary OS user here. I'm using the camera app in an old Dell Inspiron 3442 (2014) and I just got this problem.

> My guess based on what I've seen in applications like OBS, is that the app is pulling in the highest possible resolution from the camera, which is not good as that resolution runs at 2-5FPS max. Possibly having options to change the resolution in-app may help.

This comment made by @Schyken seems a good reason to get slow recording on low end machines like mine.

So, I changed the capture resolution to a lower one (320x240) in the source code here:

https://github.com/totakaro/camera/blob/fd3150cd32d0090c10312d02e415ac5c1d866ca9/src/MainWindow.vala#L48-L49

and here:

https://github.com/totakaro/camera/blob/fd3150cd32d0090c10312d02e415ac5c1d866ca9/src/MainWindow.vala#L117

And now video recording works smoothly! :tada: 

Before:
[image]

After:
[image]

A good solution would be a menu to select camera resolutions and framerates.

By example, my old laptop webcam res. looks like this:

```
$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
```

BTW, the build and install process of this `io.elementary.camera` app was straightforward,

I just followed the `README.md` Thank you so much!

cassidyjames, 2021-12-07:
@tintou's suggestion is:

>We could provide choice while still be clever and default to the highest resolution >=24fps

Which I think I agree with. It's clear from testing a wide variety of hardware that there are so many different issues with resolution and framerate, so maybe offering a (smart) choice would widen the hardware compatibility the most.