elementary/camera #38: Mirror image created eg text is reversed

bigalgeorge, 2017-08-11:
### Steps to reproduce the issue

1. Take a picture of a page and text is mirrored when it needs to be copied
2. 
3. 

### Expected behavior
Option or parameter available to allow eg mirror, or copy mode.

Describe what you expected to happen.
I use camera to take pictures of parcels before sending them and the image is a mirror, so I have to horizontal flip to fix text. I would expect to have mirror or copy sticky setting either in parameters or in UI
### Actual behavior

Describe what actually happened, being specific about how this differs from your expectations.

### System information

- **Operating system version**:

bigalgeorge, 2017-08-11:
Also, could we please have [Enter] as additional capture key

alainm23, 2018-05-27:
This could be done with a small menu that allows us to select if we want to take a picture of a document or a portrait, and play with Gdk.Pixbuf

igordsm, 2020-09-19:
Working on it! 

[image]

PR #138 (2020-11-16): Add option to mirror the webcam image
Fix #38 : add option to mirror image. 

[image]

Following the discussions on #67 , IMHO lacking this kind of feature may take users away from using camera. Specially when recording videos, being able to do simple adjustments directly from the app would make it infinitely more useful for me. I initially implemented it using a simple button, mas changed it to an app menu so that #51 could be implemented as well (I have a prototype, in fact).

What do you think? If you think its better to fix it one at a time I could change this one to use a "object-flip-hor [... cut]
closed 2020-11-16 by danrabbit