elementary/camera #41: Switch to USB webcam if connected

grvrulz, 2017-10-11:
### Steps to reproduce the issue

1. Plugin a USB webcam in a laptop
2. Open Camera application

### Expected behavior

I expected it to switch it to the USB webcam

### Actual behavior

The application keeps using the built-in webcam, and gives no indication of an external webcam

### System information

- **Operating system version**: elementary OS  0.4.1 Loki
- **Laptop*: Dell XPS 13 9350
- **Webcam**: Logitech C310

### Additional information
Both cameras work and can be switched in video calling applications like browser based services and others.

igordsm, 2020-08-29:
OK, so I now have a first implementation of this. I am still figuring out how to style the MenuToolButton, but it works. 

[image]

Besides that, I still need to figure out how to filter GStreamer inputs. For some reason each of my cameras appear twice and I even have my touchpad appearing as a camera. OBS also shows the touchpad, but manages to show only one entry for each camera.

igordsm, 2020-08-30:
It's almost there, but I am still having problems with styling the menu button. Any help is appreciated ;)

[image]

cassidyjames, 2020-08-31:
@igordsm that actually looks like a stylesheet bug which should be resolved in the new stylesheet. Let me grab some screenshots on my elementary OS 6 build.

igordsm, 2020-09-01:
@cassidyjames The current prototype is at https://github.com/igordsm/camera/tree/camera-switch .  By the way, is there a way to obtain libhandy-1 ? I can no longer compile `master`, as the newest changes rely on it.

igordsm, 2020-09-03:
@cassidyjames  Looks like you were right. It looks fine in Elementary 6:

[image]

Thank you for the heads up. By the way, should the style error be reported somewhere?

best regards.

igordsm, 2020-09-06:
Some updates on this: with the new changes for detecting devices merged the menu now displays correctly. It also shows fine in Elementary 6 (but apparently not in Hera).

1. when a USB camera is plugged the app now automatically changes to it.
2. if the current camera is unplugged the app shows the first camera.
3. menu is updated correctly.

The only problem is that taking pictures messes things up. When disposing the pipeline for the old camera some references are dangling, which causes GStreamer warnings about unreferenced data. Code is in [https://github.com/igordsm/camera/tree/camera-switch](https://github.com/igordsm/camera/tree/camera-switch) for those who want to test it. I'll open a PR as soon as I'm able to snap photos/record videos with both cameras.

PR #134 (2020-12-08): Add option to switch cameras
This PR closes #41 and closes #59 . It adds a dropdown menu with possible cameras and switches to newly plugged ones on the fly. If more than one camera is present the last one provided by `Gst.DeviceMonitor` is used. 

[image]
closed 2020-12-08 by danrabbit