elementary/camera #47: Improve Type Change Widget

alainm23, 2018-05-25:
The widget to change the type is not very intuitive, we could use a Gtk.Switch for this, I am sending a PR, to see what you think.

danrabbit, 2018-05-28:
I started working on an abstract version of the mode switch widget here: https://github.com/danrabbit/lookbook/blob/master/src/Widgets/ModeSwitch.vala

One thing I would like to solve is making sure the switch is pointing the right direction in RTL. Another nicety would be activating when clicking the icons.

I'd love any feedback on the API there

alainm23, 2018-05-29:
The code is clean and very functional, we could add this to Granite, so your application, camera and other applications could come from this.

PR #64 (2018-05-30): Use mode switch to change photo/video
Supersedes #53 and #48 
Fixes #47 

This uses the mode switch widget I wrote for LookBook

## BEFORE:

[image]

## AFTER:

[image]
closed 2018-05-30 by danrabbit