elementary/camera #50: Add a timer widget

alainm23, 2018-05-26:
A function like this is essential, especially if you want to take a selfie with friends.
This is a possible solution:
[image]

calo001, 2018-05-27:
Emojis as stickers in the camera view like WhatsApp
[image]

alainm23, 2018-05-27:
@calo001 I think that functionality has to go to another Issue.

alainm23, 2018-05-27:
I'm working on a sound for the timer, but what could this sound be?

donadigo, 2018-05-27:
@alainm23 for now use any sound for testing purposes really, just focus on the base functionality and when we will get around to review your PR, we will figure the sound out.

alainm23, 2018-05-27:
I just sent a new commit

danrabbit, 2018-05-27:
I've attached a symbolic timer icon so we don't abuse the system icons here

[timer-symbolic.svg.zip](https://github.com/elementary/camera/files/2042828/timer-symbolic.svg.zip)

alainm23, 2018-05-27:
How do I install this icon? will come by default in June or I have to create a .gresource file

danrabbit, 2018-05-28:
@alainm23 you'll need to create a gresource

danrabbit, 2018-05-28:
For sounds, it would be cool to do something like: https://www.youtube.com/watch?v=wMQzC9dNY1I

Gotta find some free sounds to use :)

PR #58 (2018-05-29): Add Timer Widget
Fixes #50
closed 2018-05-29 by danrabbit