elementary/camera #54: Keep the webcam's aspect ratio while resizing the window

eric0001, 2018-05-27:
Lock window resizing to the aspect ratio of the webcam so that there are no black borders appearing. Just like it's advertised on elementary.io 
https://elementary.io/images/screenshots/camera.jpg
On that screenshot you see no black borders on each side or on top and bottom of the window.
Even while resizing the window with the mouse, the window is locked to the canvas' aspect ratio of the webcam.
This feature also exists in audience (aka pantheon-videos) but it's broken and this feature exists in mpv-player.

DanyGee, 2018-11-04:
And the camera app should **start with its window matched to camera resolution**, just like advertised:
https://elementary.io/images/screenshots/camera.jpg

totakaro, 2019-04-24:
Hey @eric0001 and @DanyGee see my comment here: https://github.com/elementary/camera/issues/35#issuecomment-486050320