elementary/camera #80: [i18n] The string "%d Sec" seems to be displayed as untranslated

ryonakano, 2018-07-21:
### Steps to reproduce the issue

1. Open the app
2. Click the "Disabled" label at the top left of the window
3. The label changes to "3 Sec" and this seems to be displayed as untranslated

### Expected behavior

The string "%d Sec" should be displayed as translated.

### Actual behavior

[image]

The string "%d Sec" is displayed as untranslated.

### System information

- Juno Beta 1 (Japanese)

davidmhewitt, 2018-10-15:
I believe this is a bug in xgettext or gettext, because the singular and plural forms are the same for that string, it causes a conflicting ID in the po/pot files. Not really sure how to fix this one.

ryonakano, 2019-08-25:
What about changing the original string to "%s Second" (singular) and "%s Seconds" (plural)? I suppose it would fix the issue because the string for the singular and plural is not the same.

fitojb, 2019-09-27:
Or if you really intended to use a shortened form for the word “second”, just use its official SI symbol _s_. Symbols don’t take plurals.

ryonakano, 2021-05-18:
Fixed by 33c7e17

closed 2021-05-18 by ryonakano