elementary/camera #87: Camera doesn't work

Djaler, 2018-10-18:
It just print this:
`(io.elementary.camera:23548): Clutter-Gst-WARNING **: 21:17:22.625: Internal data stream error.`

### System information
- **Operating system version**: Juno daily

reedlove, 2018-10-30:
I'm having the same problem on my Dell Inspiron 15 7559.

This was not an issue in Loki.

Here's the information on my Camera:

description: Video product: Integrated_Webcam_HD vendor: CN06307G7248761BE247A03 physical id: 4 bus info: usb@1:4 version: 57.28 capabilities: usb-2.00 configuration: driver=uvcvideo maxpower=500mA speed=480Mbit/s

delacruzbordajn, 2018-11-01:
Same problem with built-in camera on Dell Inspiron 5567. I'm using Elementary OS 0.5 Juno

Error: (io.elementary.camera:19547): Clutter-Gst-WARNING **: 15:03:08.101: Internal data stream error.

This was no an issue in Loki.

Jason5679, 2018-11-08:
Camera App didn't work with HP Power 15 (15-cb004ng). Camera App show a black screen and freeze if i shot a photo. On Discord e.g. it works.

benages, 2018-11-17:
I'm having the same problem on my Acer Aspire 5749.

This is a bug in [Gstreamer](https://bugzilla.gnome.org/show_bug.cgi?id=796789), a dependency of camera. It occurs also in [Ubuntu Bionic](https://bugs.launchpad.net/ubuntu/+source/cheese/+bug/1758994).

dackdel, 2018-11-19:
Razer Blade 2017 late non-touch model. it opens but glitches like crazy its just fuzzy colours moving constantly. 
[image]

PlugaruT, 2018-11-20:
The same thing happens on a Dell XPS 13 9360.

liapman, 2018-12-11:
I have the same issue on my Dell XPS 13 9350. The camera works fine on Skype and in Chrome, but the preinstalled camera app displays a black window.

thomasvht, 2018-12-11:
I have the same issue as @liapman . 

I have an Dell Precission 5530, camera works fine in other applications (hangouts, slack, messenger,...) but in the camera app of elementary os (Juno) i got a black image and when i take a picture the program freezes.

jade28, 2018-12-21:
The same issue on HP Probook 450 G3.

slayerjain, 2019-01-06:
Same here on XPS 13 9360.

dackdel, 2019-01-07:
is this even being addressed?

PlugaruT, 2019-01-07:
I think this should be fixed upstream and not here :/

diegoleme, 2019-01-09:
Same here on Dell Inspiron 15 7560

iamzeng, 2019-01-09:
Same issue on Dell XPS 13 9370.

terrakok, 2019-01-12:
Same issue on Dell Vostro 5370:
`(io.elementary.camera:28555): Clutter-Gst-WARNING **: 16:52:12.835: Internal data stream error.
`

LucasAsafe, 2019-01-20:
[image]
Same problem reported above. I got this black image when i open the camera app, cheese not working too. Dell inspiron 7572

Jetman823, 2019-01-28:
Alienware 15 r3, same problem as the one above me. Black screen, if you hit the capture button it crahses the program.

hannesschulze, 2019-02-08:
Same problem on the Huawei Matebook X Pro

mrsfigatto, 2019-02-17:
Same here with XPS 13 9370....

PR #93 (2020-07-07): Remove Clutter support, use native Gtk element from GStreamer
- [x] Add photo taking support (to JPG)
- [x] Add video recording support (to OGV ?)
Fixes #87
closed 2020-07-07 by danrabbit