elementary/camera #89: GStreamer error: negotiation problem

dskrad, 2019-01-03:
### Steps to reproduce the issue

Open camera app via terminal `io.elementary.camera`

### Expected behavior

Camera should load live image and be able to take video or snapshot.

### Actual behavior

Window is completely black. Terminal outputs the following warning:
`(io.elementary.camera:10892): Clutter-Gst-WARNING **: 10:46:09.371: GStreamer error: negotiation problem.`

### System information

- elementary Juno

cassidyjames, 2019-01-03:
@dskrad does using your webcam with the app Cheese (from AppCenter) work?

dskrad, 2019-01-03:
I just downloaded Cheese and it cannot find my webcam.  VLC can find my webcam no problem.

Here are the errors I get with Cheese:

    ** Message: 12:07:55.709: cheese-application.vala:211: Error during camera setup: No device found

    (cheese:17873): cheese-CRITICAL **: 12:07:55.726: cheese_camera_device_get_name: assertion 'CHEESE_IS_CAMERA_DEVICE (device)' failed

    (cheese:17873): GLib-CRITICAL **: 12:07:55.726: g_variant_new_string: assertion 'string != NULL' failed

    (cheese:17873): GLib-CRITICAL **: 12:07:55.726: g_variant_ref_sink: assertion 'value != NULL' failed

    (cheese:17873): GLib-GIO-CRITICAL **: 12:07:55.726: g_settings_schema_key_type_check: assertion 'value != NULL' failed

    (cheese:17873): GLib-CRITICAL **: 12:07:55.726: g_variant_get_type_string: assertion 'value != NULL' failed

    (cheese:17873): GLib-GIO-CRITICAL **: 12:07:55.726: g_settings_set_value: key 'camera' in 'org.gnome.Cheese' expects type 's', but a GVariant of type '(null)' was given

    (cheese:17873): GLib-CRITICAL **: 12:07:55.726: g_variant_unref: assertion 'value != NULL' failed

    ** (cheese:17873): CRITICAL **: 12:07:55.726: cheese_preferences_dialog_setup_resolutions_for_device: assertion 'device != NULL' failed

On elementary Loki, the camera or cheese apps found my webcam right away.

cassidyjames, 2021-12-06:
@dskrad given that, it sounds like it might be a kernel or GStreamer issue, and not specific to the camera app. I'd be curious if the newer versions of the camera app works on your hardware; you can test the Flatpak here: https://appcenter.elementary.io/io.elementary.camera/

I'm going to close this issue as I don't think there's anything specific for us to do here, but we can always re-open if we can identify it as a solvable issue within this codebase.

closed 2021-12-06 by cassidyjames