elementary/camera #90: Stock Camera App  doesn't work on Del Vostro 15 series

thoughtoinnovate, 2019-01-15:
### Steps to reproduce the issue

1. Install elementary juno on dell vostro 15 laptop
2.  open stock "camera" application
3. Camera doesn't work

### Expected behavior

Stock Camera app should work , camera light show light up.

### Actual behavior

Dark screen on camera app , camera light doesn't blink

### System information

elementary Juno

savegame, 2019-09-11:
Yes, have Dell Vostro 5, stok camera app still not work.

ryonakano, 2019-09-11:
I suppose this issue is duplicate of #103.

@thoughtoinnovate @savegame Could either of you open Terminal, type and run the following command, and paste the output here?

```bash
io.elementary.camera
```

savegame, 2019-09-12:
by SSH (laptop at home):
DISPLAY=:0 io.elementary.camera 
(io.elementary.camera:5663): Clutter-Gst-CRITICAL **: 08:47:03.628: Unable to setup device without element factory, node and name set 0x5653540ef0e0 0x7fb60800e0f0 (nil)

ryonakano, 2019-09-13:
@savegame Judging from that error message, this issue is the same issue with #103, so I'm closing this one as a duplicate of that one.

closed 2019-09-13 by ryonakano