elementary/camera #92: The io.elementary.camera package installs on Ubuntu Disk 19.04 normally, but does not open the video image.

serial02, 2019-03-08:
Hello Team Elementary OS

Can you help me with this?

https://answers.launchpad.net/ubuntu/+source/camera-app/+question/679039

edson@edson-p6540br:~$ uname -a
Linux edson-p6540br 5.0.0-7-generic #8-Ubuntu SMP Mon Mar 4 16:27:25 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
edson@edson-p6540br:~$

*************

edson@edson-p6540br:~$ lsb_release -crid
Distributor ID:	Ubuntu
Description:	Ubuntu Disco Dingo (development branch)
Release:	19.04
Codename:	disco

*************
As for the command: LANG = C sudo dpkg --audit did not appear anything, simply typed the password and did not list anything.

Look:

edson@edson-p6540br:~$ LANG=C sudo dpkg --audit
[sudo] password for edson:

*************

edson@edson-p6540br:~$ LANG=C apt-cache policy io.elementary.camera pantheon-camera snap-photobooth
io.elementary.camera:
  Installed: 1.0.3+r718+pkg15~daily~ubuntu18.10.1
  Candidate: 1.0.3+r718+pkg15~daily~ubuntu18.10.1
  Version table:
 *** 1.0.3+r718+pkg15~daily~ubuntu18.10.1 500
        500 http://ppa.launchpad.net/elementary-os/daily/ubuntu disco/main amd64 Packages
        100 /var/lib/dpkg/status
pantheon-camera:
  Installed: (none)
  Candidate: 1.0.3+r718+pkg15~daily~ubuntu18.10.1
  Version table:
     1.0.3+r718+pkg15~daily~ubuntu18.10.1 500
        500 http://ppa.launchpad.net/elementary-os/daily/ubuntu disco/main amd64 Packages
snap-photobooth:
  Installed: (none)
  Candidate: 1.0.3+r718+pkg15~daily~ubuntu18.10.1
  Version table:
     1.0.3+r718+pkg15~daily~ubuntu18.10.1 500
        500 http://ppa.launchpad.net/elementary-os/daily/ubuntu disco/main amd64 Packages
edson@edson-p6540br:~$

********************
When you open the camera through the terminal the following error message appears:

edson@edson-p6540br:~$ io.elementary.camera

(io.elementary.camera:16631): Clutter-WARNING **: 09:27:57.062: Whoever translated default:LTR did so wrongly.

(io.elementary.camera:16631): Clutter-Gst-CRITICAL **: 09:27:57.166: Unable to setup device without element factory, node and name set 0x55c94c5293c0 0x7fc58c00dd00 (nil)

**********************************

https://bugs.launchpad.net/ubuntu/+source/camera-app/+bug/1818902

Hello guys

I installed the io.elementary.camera package from Elementary OS on Ubuntu Disk 19.04 normally, but message appears that it does not recognize the webcam. Note I have the Juno Elementary OS installed on the same machine and the Webcam works normally.

I installed through the repository:

sudo add-apt-repository ppa: elementary-os / daily
sudo apt update
sudo apt install io.elementary.camera

In the description, there is the option pantheon-camera, but this one says to be with broken package and does not install. Could it be this that does not let the webcam work? Look!

sudo apt install pantheon-camera
[sudo] password for edson:
Reading package lists ... Ready
Building dependency tree
Reading status information ... Ready
Some packages could not be installed. This may mean that
you asked for an impossible situation or, if you are using the
unstable distribution, that some required packages have not been created yet or have been removed from Incoming.
The following information may help resolve the situation:

The following packages have mismatched dependencies:
pantheon-camera: Depends: io.elementary.camera but will not install
E: Impossible to fix problems, you kept (hold) broken packages.

Note When I enter synaptcis there is no broken package.

If you uninstall io.elementary.camera and try to install the pantheon-camera the same broken package information appears.

The installation of io.elementary.camera is performed successfully on Ubuntu 19.04 Disk, but does not appear the image of the video and nor does it recognize the webcam, video attachments.

When you try to open the camera through the terminal the following error message appears:

edson @ edson-p6540br: ~ $ io.elementary.camera

(io.elementary.camera:10218): Clutter-WARNING **: 16: 43: 29.698: Whoever translated default: LTR did so wrongly.

(io.elementary.camera:10218): Clutter-Gst-CRITICAL **: 16: 43: 29.753: Unable to setup device without element factory, node and name set 0x55a19aa51200 0x7fc2ec00e240 (nil)
edson @ edson-p6540br: ~ $

I ask you to forward the responsible team so that the correction can be carried out.

I thank you for your attention

Serial.com

Note Even after updating the version on today's date, the io-elementary-camera does not work on Ubuntu 19.04 Disk.

É preciso baixar 36,9 kB de arquivos.
Depois desta operação, 0 B adicionais de espaço em disco serão usados.
Obter:1 http://ppa.launchpad.net/elementary-os/daily/ubuntu disco/main amd64 io.elementary.camera amd64 1.0.3+r718+pkg15~daily~ubuntu19.04.1 [36,9 kB]
Baixados 36,9 kB em 1s (26,8 kB/s)              
(Lendo banco de dados ... 313409 ficheiros e directórios actualmente instalados.)
A preparar para desempacotar .../io.elementary.camera_1.0.3+r718+pkg15~daily~ubuntu19.04.1_amd64.deb ...
A descompactar io.elementary.camera (1.0.3+r718+pkg15~daily~ubuntu19.04.1) sobre (1.0.3+r718+pkg15~daily~ubuntu18.10.1) ...
Configurando io.elementary.camera (1.0.3+r718+pkg15~daily~ubuntu19.04.1) ...
A processar 'triggers' para desktop-file-utils (0.23-4ubuntu1) ...
A processar 'triggers' para mime-support (3.60ubuntu1) ...
A processar 'triggers' para gnome-menus (3.31.4-2ubuntu2) ...
A processar 'triggers' para libglib2.0-0:amd64 (2.60.0-1) ...
override for key “transparency-mode” in schema “org.gnome.shell.extensions.dash-to-dock” in override file “/usr/share/glib-2.0/schemas/10_ubuntu-dock.gschema.override” is not in the list of valid choices; ignorando sobrescrita para esta chave.
A processar 'triggers' para bamfdaemon (0.5.3+18.04.20180207.2-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Lendo listas de pacotes... Pronto
Construindo árvore de dependências       
Lendo informação de estado... Pronto
Calculando atualização... Pronto
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 0 não atualizados.
Lendo listas de pacotes... Pronto
Construindo árvore de dependências       
Lendo informação de estado... Pronto
Calculando atualização... Pronto
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 0 não atualizados.
Lendo listas de pacotes... Pronto
Construindo árvore de dependências       
Lendo informação de estado... Pronto
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 0 não atualizados.
edson@edson-p6540br:~$ io.elementary.camera

(io.elementary.camera:4241): Clutter-WARNING **: 11:47:25.308: Whoever translated default:LTR did so wrongly.

(io.elementary.camera:4241): Clutter-Gst-CRITICAL **: 11:47:25.368: Unable to setup device without element factory, node and name set 0x556a5d8eb3c0 0x7fe59800dd00 (nil)
edson@edson-p6540br:~$

[image]
[image]

serial02, 2019-03-11:
Up

fabiozaramella, 2019-08-08:
I'll close this a a duplicate of #103  
But thanks for all the info you provided.

closed 2019-08-08 by fabiozaramella

savegame, 2019-09-11:
still not work

ryonakano, 2019-09-11:
@savegame The issue is fixed in #106, but the new release of Camera that includes this fix has not been released yet.