elementary/camera #99: Add configuration option for switching resolution (Logitech cameras)

itsmingjie, 2019-05-12:
### Steps to reproduce the issue

Currently using a Logitech C615 webcam.

### Expected behavior

Logitech C615 supports 2 recording resolution/modes, both wide and standard. I am able to record in 16:9 in other applications such as Zoom, or even the browser. It'd be great if we can have an option to switch the resolution.

### Actual behavior

Currently the Camera app is only giving me a 4:3 video without any options to change it to wide.

### System information

- **Operating system version**: Juno

carniz, 2019-09-25:
Version 1.0.5 restricts the resolution to 640x480 (see https://github.com/elementary/camera/releases/tag/1.0.5), but it would be great to make it possible to switch resolution - regardless of which camera is being used.

I'm also suggesting that it could default to the highest possible resolution supported by the camera instead of locking it to 640x480, since most cameras today support much higher resolutions than 640x480.

closed 2020-06-16 by itsmingjie

jtrees, 2020-08-10:
Why was this closed?