elementary/camera #103: Clutter-Gst-CRITICAL

jacob1237, 2019-07-01:
### Overview

After the latest update (or after I tried to install `xserver-xorg-core-hwe-18.4`), the camera app stopped working. Previously, I was able to use it normally.

The camera works fine under Skype, Cheese and VLC.

### Expected behavior

The camera app must detect my default available camera.

### Actual behavior

When I open the app, it prints the error in the main window: `No Supported Camera Found. Connect a webcam or other supported video device to take photos and video.`

Also, I see the following console error:
```
(io.elementary.camera:29113): Clutter-Gst-CRITICAL **: 14:19:09.325: Unable to setup device without element factory, node and name set 0x56461611fc60 0x7f297800c810 (nil)
```

### System information

Acer Swift 3 315-52G UHD

- **Operating system version**: Juno with latest updates

darsain, 2019-07-01:
Same issue here. I always delay updates because it always means something is gonna break and I'll need to troubleshooting new issues, but after like a month I finally decided to click on the update button aaand camera doesn't work :/

Lenovo Yoga 500

4jNsY6fCVqZv, 2019-08-02:
@danrabbit This seems to be a widespread issue.
https://github.com/elementary/camera/issues/103
https://github.com/elementary/camera/issues/104
In this state the app unfortunately doesn't work anymore.
Can you please provide a solution?

4jNsY6fCVqZv, 2019-08-08:
@fabiozaramella Thank you. Will you look into it?

Fatih20, 2019-08-09:
Same issue, I'm using a thinkpad x230.

roypen, 2019-08-10:
The same issue on Acer E1-571 :

`(io.elementary.camera:2048): Clutter-Gst-CRITICAL **: 20:11:11.226: Unable to setup device without element factory, node and name set 0x55eac557bbe0 0x7f15a000dad0 (nil)`

I tried daily build but with no success.

roypen, 2019-08-11:
It looks like there is a problem with 

```
if (camera_devices.length > 0 && **camera_devices[0].get_name () != null**) {
                initialize_camera_view ();
            } else {
                stack.set_visible_child_name ("no-device");

                debug ("No camera device found.");
            }

            return false;
```
Somehow my Chicony camera name is null. Removing `camera_devices[0].get_name () != null` from code solves the problem and webcam is finally working!

cassidyjames, 2019-08-12:
@roypen good catch! Apparently it's the same on my Dell Precision as well.

closed 2019-08-14 by cassidyjames

jlamasfripp, 2019-09-22:
Same here! Desktop pc, installed today. 
[image]

ryonakano, 2019-09-23:
@jlamasfripp The issue was fixed in #106 but it remains on your system until a new version of Camera is released.

PR #106 (2019-08-14): Handle null camera name
Fixes #103

### Changes Summary

- removing ` && camera_devices[0].get_name () != null`

My webcam name is "null", that's why the app wasn't working. After removing this part of code everything works fine.