elementary/camera #104: camera not detected

wurkin, 2019-07-09:
### Steps to reproduce the issue

1. using Lenovo Thinkpad x220
2. click on bundled camera app
3. camera not detected

### Expected behavior

camera would turn on
Describe what you expected to happen.

### Actual behavior
camera is not found
Describe what actually happened, being specific about how this differs from your expectations.

### System information

- **Operating system version**:

After running: `hwinfo -usb`

` P: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0
  E: DEVPATH=/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0
  E: DEVTYPE=usb_interface
  E: DRIVER=uvcvideo
  E: ID_MODEL_FROM_DATABASE=Lenovo Integrated Camera (0.3MP)
  E: ID_USB_CLASS_FROM_DATABASE=Miscellaneous Device
  E: ID_USB_PROTOCOL_FROM_DATABASE=Interface Association
  E: ID_VENDOR_FROM_DATABASE=Chicony Electronics Co., Ltd
  E: INTERFACE=14/1/0
  E: MODALIAS=usb:v04F2pB217d0854dcEFdsc02dp01ic0Eisc01ip00in00
  E: PRODUCT=4f2/b217/854
  E: SUBSYSTEM=usb
  E: TYPE=239/2/1
  E: USEC_INITIALIZED=31094031`

after that i initialized the camera with `modprobe uvcvideo`
installed cheese camera app `sudo apt-get install cheese`
ran `cheese`

And it worked.

wurkin, 2019-07-09:
Even after inititalization with modprobe the bundled camera app did not work.

ryonakano, 2019-07-15:
@wurkin  Could you open Terminal, type and run the following command, and paste the output here?

```bash
io.elementary.camera
```

fabiozaramella, 2019-08-08:
I'll close this as a duplicate of #103

closed 2019-08-08 by fabiozaramella