elementary/camera #105: My webcam is not detected in this app

lohxx, 2019-07-14:
### Steps to reproduce the issue

1. Open the app. 

### Expected behavior

Is that my webcam be detected

### Actual behavior

My webcam is not being detected.

### System information

- **Operating system version**:  elementary OS 5.0 Juno, this happens in zenbook 14 UX433FA

ryonakano, 2019-07-15:
@lohxx  Could you open Terminal, type and run the following command, and paste the output here?

```bash
io.elementary.camera
```

lohxx, 2019-07-16:
➜  ~ io.elementary.camera

(io.elementary.camera:3677): Clutter-WARNING **: 23:07:06.753: Whoever translated default:LTR did so wrongly.

(io.elementary.camera:3677): Clutter-Gst-CRITICAL **: 23:07:06.778: Unable to setup device without element factory, node and name set 0x5572cfc3c680 0x7f000000c810 (nil)

ryonakano, 2019-07-17:
Hmm it looks like this issue may be duplicate of #103 as far as judging from that log.

fabiozaramella, 2019-08-08:
@ryonakano It is. I'll close this as a duplicate of #103

closed 2019-08-08 by fabiozaramella