elementary/camera #113: There's no feedback when a video is recorded

alainm23, 2019-12-22:
When the video is recorded successfully, there's no no way to tell the user that the video was recorded correctly and where it was saved.

### Solution
We can send a notification to the user indicating that their video was saved correctly and if the user clicks on the notification, that the notification opens the file browser where the video was saved.

cassidyjames, 2020-10-28:
A notification isn't the right solution here. Notifications should be used when the app is not focused or visible; if the app is being used, an in-app toast would make more sense.

I also want to clarify that the purpose of a toast is not that the action completed successfully—we should always assume success and only alert on errors—but to provide a handy way to get to the recorded file.

So a toast like this might make sense:

>Saved to Videos `View File`

Thoughts from @elementary/ux?

igordsm, 2020-11-13:
@cassidyjames Thank you for the explanations. How about the following design? 

[image]

cassidyjames, 2020-11-13:
@igordsm I'm happy with that!

PR #142 (2021-05-14): Send notification when recording has finished.
Closes #113 .

[image]

This is very open for comments. I had never used notifications and actions before, so this may not be the best or even correct way of doing this. If this is indeed garbage, I would appreciate some pointers on how to do this right. 

Best regards,
closed 2021-05-14 by jeremypw