elementary/camera #117: io.elementary.camera do not working

omarcoscardoso, 2020-04-03:
my notebook is eOS 5.2.1 Hera, 

my notebook it is lenovo g400s, eOS 5.2.1 Hera.
I open camera and is not working, is all black.

[image]

my webcam is working in other applications, like cheese

alfianhanafi, 2020-04-17:
have the same issue, well i'm rarely use camera, so i don't know exactly when it's not working.

when i first installed Elementary OS, the camera it's not working on my laptop.
then you make an update in September, the camera working --> [Elementary Blog - Sep. update](https://blog.elementary.io/updates-for-september-2019/)
now it's not working again...

the output from terminal is exactly like above
(io.elementary.camera:17739): Clutter-Gst-CRITICAL **: 11:44:57.109: Unable to setup device without element factory, node and name set 0x55ee6e33dee0 0x7f1b0000dd40 (nil)

thanks

abccyz, 2020-05-26:
> 我的笔记本是eOS 5.2.1 Hera，
> 
> 我的笔记本是lenovo g400s，eOS 5.2.1 Hera。
> 我打开相机，无法正常工作，全黑。
> 
> [image]
> 
> 我的网络摄像头正在其他应用程序中工作，例如奶酪
> 
> 是返回终端...。
> 
> （io.elementary.camera:5360）：Clutter-Gst-CRITICAL **：11：46：12.675：在没有元素工厂，节点和名称设置的情况下无法设置设备0x56182d6b40f0 0x7f153400b950（无）
> 
> 要支持此问题吗？**[发布赏金吧！](https://www.bountysource.com/issues/90865371-io-elementary-camera-do-not-working?utm_campaign=plugin&utm_content=tracker%2F45629460&utm_medium=issues&utm_source=github)**我们通过[Bountysource](https://www.bountysource.com/?utm_campaign=plugin&utm_content=tracker%2F45629460&utm_medium=issues&utm_source=github)接受赏金。

How do you use 5.2.1 OS? I can only see 5.1.2 here

abccyz, 2020-05-26:
How do you use 5.2.1 OS? I can only see 5.1.2 here

omarcoscardoso, 2020-05-26:
Desculpe, eu queria dizer 5.1.2

cassidyjames, 2021-12-06:
The preview has been re-written; can you give the camera on elementary OS 6 a try, or try the latest release as a Flatpak from here: https://appcenter.elementary.io/io.elementary.camera/

closed 2021-12-06 by cassidyjames