elementary/camera #119: caera issue

robin113x, 2020-04-12:
hey ,
i'm using elementry os from few year ago but from few weeks  i have lot's of issues  such that camera and bluetooth was not open . please fix this problem,

thank you

ryonakano, 2020-04-12:
Could you open Terminal, run the following command, and paste the output here?

```
io.elementary.camera
```

t-io, 2020-05-07:
After a reinstallation on my XPS the camera stopped working.
`io.elementary.camera` returns

(io.elementary.camera:17640): Clutter-Gst-CRITICAL **: 21:26:50.214: Unable to setup device without element factory, node and name set 0x5562b2ca70a0 0x7f0b5000faf0 (nil)

cassidyjames, 2021-12-06:
@k0hinoor since this is a pretty broad issue report and there has been no response to follow-up questions, I'm going to close it. If you can provide more info, we can consider re-opening.

closed 2021-12-06 by cassidyjames