elementary/camera #120: Add the drivers for a MAC Book facetimehd camera

ralphsc, 2020-04-24:
I would like to propose the the facetimehd drivers be added when installing on a MAC Book.
Currently I have to run this script to install the drivers:

#!/bin/bash
set -e

#export CONFIG_MODULE_SIG=n
#export CONFIG_MODULE_SIG_ALL=n
#export KERNELRELEASE=${1}

echo "Installing FacetimeHD camera for $KERNELRELEASE"
cd /tmp
git clone https://github.com/patjak/bcwc_pcie.git
cd bcwc_pcie/
git clone https://github.com/patjak/facetimehd-firmware.git
cd facetimehd-firmware/
make
make install
cd ..
make
make install
rm -rf /tmp/bcwc_pcie

if [ ! -d "/etc/modules-load.d" ]; then
  mkdir -p "/etc/modules-load.d"
fi

cat > "/etc/modules-load.d/facetimehd.conf" << EOL
videobuf2-core
videobuf2_v4l2
videobuf2-dma-sg
facetimehd
EOL

echo "Running depmod"
depmod -a

echo "Install complete."

StaticVengeance, 2020-06-18:
> I would like to propose the the facetimehd drivers be added when installing on a MAC Book.
> Currently I have to run this script to install the drivers:
> 
> #!/bin/bash
> set -e
> 
> #export CONFIG_MODULE_SIG=n
> #export CONFIG_MODULE_SIG_ALL=n
> #export KERNELRELEASE=${1}
> 
> echo "Installing FacetimeHD camera for $KERNELRELEASE"
> cd /tmp
> git clone https://github.com/patjak/bcwc_pcie.git
> cd bcwc_pcie/
> git clone https://github.com/patjak/facetimehd-firmware.git
> cd facetimehd-firmware/
> make
> make install
> cd ..
> make
> make install
> rm -rf /tmp/bcwc_pcie
> 
> if [ ! -d "/etc/modules-load.d" ]; then
> mkdir -p "/etc/modules-load.d"
> fi
> 
> cat > "/etc/modules-load.d/facetimehd.conf" << EOL
> videobuf2-core
> videobuf2_v4l2
> videobuf2-dma-sg
> facetimehd
> EOL
> 
> echo "Running depmod"
> depmod -a
> 
> echo "Install complete."
> 
> Want to back this issue? **[Post a bounty on it!](https://www.bountysource.com/issues/91429169-add-the-drivers-for-a-mac-book-facetimehd-camera?utm_campaign=plugin&utm_content=tracker%2F45629460&utm_medium=issues&utm_source=github)** We accept bounties via [Bountysource](https://www.bountysource.com/?utm_campaign=plugin&utm_content=tracker%2F45629460&utm_medium=issues&utm_source=github).

I can never get the my Macbook Pro 2017 camera to work with Cheese or the Camera app. It works with mplayer and Skype. Just tried your script still no luck.

ralphsc, 2020-06-18:
The Camera app worked with other kernels but stopped working two kernels
ago. The script does make the camera work on other apps like the ones you
mentioned and also on HouseParty and Zoom.

danrabbit, 2022-06-03:
Closing because we use the Ubuntu hardware enablement stack, so any requests for hardware support should be implemented upstream in Ubuntu

closed 2022-06-03 by danrabbit