elementary/camera #122: Unable to get available camera devices, v4l2src element missing

WhyNotHugo, 2020-06-06:
## Prerequisites
- [x] I have searched open and closed issues for duplicates. (0 matches for this error message)

## Describe the bug

Application crashes at startup:

```console
$ io.elementary.camera

(io.elementary.camera:54961): Clutter-Gst-WARNING **: 14:14:23.319: Unable to get available camera devices, v4l2src element missing
zsh: segmentation fault (core dumped)  io.elementary.camera
```

### Expected behavior

Should not crash? Or maybe explain what's wrong with my setup.

### Logs

Output of `strace io.elementary.camera 2> camera.err` (github won't let me attach this file "We don't support that file type", so [here's a gist](https://gist.github.com/WhyNotHugo/72b3103c181c826294691ff4337c5870) instead.

## Platform Information
 - OS: ArchLinux
 - OS Version: amd64
 - Hardware info: `04f2:b604 Chicony Electronics Co., Ltd Integrated Camera (1280x720@30)`

- [ ] I'm using the latest version from git that I've manually compiled
- [x] I'm using the latest released stable version

## Additional context

Camera works fine on Firefox, Zoom and some other apps, so there should be driver/hardware issues with it.

tintou, 2020-06-08:
You need `gst-plugins-good`, which should be installed alongside Camera when installing from packaging

closed 2020-06-08 by tintou

WhyNotHugo, 2020-06-08:
Ah, thanks for pointing that out. I'll report the missing dependency to the package maintainer for my disto.