elementary/camera #149: The camera does not work [2015 MacBook Air 13"]

adityadasgupta, 2020-11-25:
I have elementaryOS Hera on a partition on my MacBook Air. Everything seems to work fine except the front camera. Let me know if there is a fix to this!

ilDon, 2021-01-21:
Same exact problem here, anybody has a (working) solution?

quenyan, 2021-05-07:
Same here, Macbook Air 13" Mid 2013. pitch black, no image are shown when simple camera app fired up. Not using it at this moment tho, but it's better when I know it's working.

mpanhans, 2021-06-22:
This issue persists in eOS 6 beta2. I am on a 2015 MacBook Pro 13''. I've gotten the camera to work with mplayer, Cheese, and Jitsi Meet calls, but the Camera App only shows a black screen, no image.

cassidyjames, 2021-12-03:
@adityadasgupta @ilDon @quenyan @mpanhans can you help me out: when you hit the capture button, does a light come on and is a photo actually saved?

mpanhans, 2021-12-04:
> @adityadasgupta @ilDon @quenyan @mpanhans can you help me out: when you hit the capture button, does a light come on and is a photo actually saved?

Let me first give some context. The webcam on several MacBook Pro and Air models, including 2015 models, does not work out of the box on Linux. [Patjak wrote a driver](https://github.com/patjak/facetimehd/wiki) that can be installed, after which some applications will properly work with the webcam (e.g, Jitsi, mplayer) on elementary. 

However, even after installing that facetimehd driver, the elementary Camera app does not recognize the webcam. When you open Camera, is just shows a black window and the webcam light does not turn on. If you click capture, a file is saved, but it is empty.

ilDon, 2021-12-04:
I can confirm the same behavior, thanks for looking into this!

danirabbit, 2025-03-03:
Closing as out of scope since this since to be a driver issue and drivers are handled upstream in the Linux kernel.

closed 2025-03-03 by danirabbit