elementary/camera #155: No feedback when no camera is connected

danrabbit, 2021-04-05:
## What Happened

I'm on a desktop computer with no connected camera. When opening the Camera app, I just see a spinner with no further explanation forever. It gives the impression that I should be waiting for the Camera to complete whatever it's doing

## Expected Behavior

There should be some kind of further feedback letting the user know that they must connect a camera and that no cameras are currently available.

igordsm, 2021-05-06:
It seems that support was in the code, but there was a missing `.show ()` in `no_device_view`. Now I get this when opening Camera without a webcam:

[image]

If this is OK I can send a PR with the (very simple) fix.

ryonakano, 2021-05-06:
@igordsm Glad if you would send a PR :blush:

igordsm, 2021-05-06:
Sure. I'll just wrap up what I've been experimenting with to fix #72 and will send a PR addressing both.

PR #159 (2021-05-14): When no camera is present show view and disable all controls.
This fixes #155 and #72 . When no camera is present this PR shows the adequate view and disables all controls. When a camera is plugged it enables all controls and shows it. 

[image]
closed 2021-05-14 by jeremypw