elementary/camera #167: hello so i have sggestion please add a feature to turn the light off in the camera app cuz i started using the usb camera and its led is on and wont even turn off so it would be helpful if i could turn offf the lights through the camera app thankyou

nooblinuxuser17, 2021-06-20:
## Problem

## Proposal

## Prior Art

danrabbit, 2021-06-20:
Hey thanks for filing a report! It would be really helpful if you could fill out the issue template and keep the title to a single sentence summary of the feature request.

Could you also please provide a bit more information about the camera you are using here? Is this LED light a colored indicator that the camera is on like a small red or green light? Is this a larger light meant to fill the subject like a ring light? Or is this meant to be more like a flash? Providing a link to a product description online of the camera would be helpful as well

cassidyjames, 2021-12-06:
Closing due to no response. @nooblinuxuser17 if you can answer the above questions, we can re-open.

closed 2021-12-06 by cassidyjames