elementary/camera #169: Enable full resolution image capture

ianmjones, 2021-07-08:
## Problem

My webcam (a Razor Kiyo) can work at the following resolutions:

* 1080p resolution (1920x1080) video at 30 fps
* 720p resolution (1280x720) video at 60 fps
* 2688x1520 still images

The Camera app however restricts photos to 640x480 resolution.

I took this picture on Elementary 5.1.17 Hera with Camera 1.0.6, and as beautiful as it is, it's not the 2688x1520 resolution I hoped for! :scream: 

[image]

## Proposal

Please enable full resolution by default with the option to switch to compatibility mode.

_or_

Enable switching from a default 640x480 compatibility mode to a full resolution mode.

If the reason for restricting the resolution to 640x480 was related to the preview being too large to fit on the screen, it's perfectly reasonable to scale and restrict the preview size while taking full resolution images.

## Notes

I'm aware that in [release 1.0.5](https://github.com/elementary/camera/releases/tag/1.0.5) the resolution was intentionally restricted to 640x480, but I propose we explore solutions that allow for using the resolutions that are available, especially when taking still images.

Issue #99 is somewhat related, but for whatever reason has been closed.

cassidyjames, 2021-07-08:
@ianmjones thanks for filing this; I guess we never followed up with this issue. I recall the problem being that the preview would get very stuttery and unusable with higher resolutions, thus we capped it in the same way as GNOME's camera app, Cheese. However, that's obviously not ideal and I'd love to see full resolution image captures return.

#99 looks to have been closed by the original issue reporter without comment, but it was asking specifically for configurable resolutions while this issue is seeking full resolution capture—related, but distinct enough to remain separate in my opinion.

igordsm, 2021-07-13:
Hey, I can work on this. I might now how to solve the stutter preview and still capture full resolution images. I'll keep you posted on my investigations here.

igordsm, 2021-07-24:
So, I've got some results in this:

1. image and video capture are different things and apparently you can only have one active at a time
2. resolutions are different, so some cameras are capable of capturing still images in much larger resolution than video
3. video resolutions do not always have the same framerate, so a camera can output VGA at 30fps and 1080p at 15 fps

I see two possible solutions for this:

1. run the preview at VGA and, when capturing images use full resolution. Of course, then capturing full resolution videos could become an issue as well
2. keep the current code and run the preview at the highest video resolution the camera supports, even if the FPS drops. If needed, there could be an option to select capture resolutions for each camera.

I'm not sure 1 is possible, but 2 is relatively easy to do. I've got some smooth preview capturing 1080p inputs, so maybe this is not an issue anymore? I'll keep investigating to see if 1 is possible, as it seems to be a good solution.

igordsm, 2021-07-25:
So 1 is possible: I'm able to preview in VGA and then take pictures in 720p. The only problem is that this creates a little "pause" effect in the preview, as shown below.

[image]

This happens because I need to set the preview pipeline to `NULL` before setting a different resolution and mode (*image* vs *video* mode). If this is acceptable I can start formatting a PR. Some questions:

1. Should we offer a way for the user to choose the picture resolution?
2. Should we just select the largest resolution and use it as default for pictures? This might have a different aspect ratio from the preview and may be confusing for the user. 

Any thoughts @cassidyjames @ianmjones ?

cassidyjames, 2021-07-28:
@igordsm thanks for digging into this! I think the pause is acceptable for a higher resolution. I agree the aspect ratio issue might be strange; I wonder if we can crop the preview to match what the captured aspect ratio would be, and if that would even be the correct thing to do… e.g. if my camera can capture 1920×1080, it's probably cropping down when previewing 640×480, so cropping the preview to 16:9 would mean we'd have far less in frame in the preview than in the captured image…

cassidyjames, 2021-07-28:
@igordsm having a user-selectable resolution may just be a concession we have to make considering the wide variety of cameras and capabilities. We could still default to the lower resolution for compatibility sake, but a user could experiment with selecting a higher one?

igordsm, 2021-07-28:
Barring some pretty obvious spacing/margin problems, would something like this be ok for selecting picture resolution?

[image]

The preview and picture capture resolution are not linked (in the code I wrote to test this), so we can theoretically still have a smooth VGA preview and only change resolution the capturing the image (this is why the preview gets stuck). How does that sound, @cassidyjames ? I would capture still picture on the maximum supported resolution by default, since this is what @ianmjones was expecting. Am I correct?

There's still a lot of refactoring to do so that mirroring and image settings get applied correctly, but it seems like I have done all the research to get this working.

igordsm, 2021-07-29:
Another option would be something similar to the display resolution popover

[image]

cassidyjames, 2021-07-29:
@igordsm a combo box like Displays would probably be the easiest overall, but I'm not sure I like that pattern. Do we know generally how many resolutions will be listed here?

igordsm, 2021-07-30:
@cassidyjames My camera reports these image formats

* image/jpeg 1280 x 720
* image/jpeg 960 x 540
* image/jpeg 848 x 480
* image/jpeg 640 x 480
* image/jpeg 640 x 360
* image/jpeg 424 x 240
* image/jpeg 320 x 240
* image/jpeg 320 x 180
* image/jpeg 160 x 120

and these video formats:

* video/x-raw 1280 x 720
* video/x-raw 960 x 540
* video/x-raw 848 x 480
* video/x-raw 640 x 480
* video/x-raw 640 x 360
* video/x-raw 424 x 240
* video/x-raw 320 x 240
* video/x-raw 320 x 180
* video/x-raw 160 x 120

This camera has matching image/video formats, but OP's camera does not so this is not something we can rely on. Most of the resolutions are also kind of nonsense. I mean, who would use 160x120 in 2021? IMHO we should either filter for common formats or just take the highest resolution possible for pictures. 

For videos it might make sense to let the user chose the resolution, since FPS can also vary. For instance, one of my cams can do 1080p, but only at 10fps.

cassidyjames, 2021-07-30:
@igordsm I agree:

- highest available resolution makes sense for photos
- we should filter some of the lowest resolutions out; maybe a minimum resolution in either dimension of 480?
- choosing resolution for video makes sense
- we should show video resolution as something like `1920×1080 60fps` if possible

I'd like to clean up resolutions and FPS even more if possible but I'm not sure how much effort it's worth and how standardized things are. E.g. I could see a list like this making sense to me:

- 1080p FHD 10fps
- 720p HD 30fps
- 480p SD 60fps

…but I don't know what to do about all the other resolutions and aspect ratios—or what to do if multiple supported FPS exist.

cassidyjames, 2021-07-30:
For prior art, my Pixel camera has two supported video resolutions: 3840×2160 and 1920×1080, and 30fps or 60fps. It shows them separately like this:

- Resolution
   - 4K (Ultra-high resolution)
   - Full HD (1080p)
- Frames/sec
   - Auto
   - 30
   - 60

I have no idea what "Auto" fps does. If you choose certain combinations of features that are incompatible, it marks the incompatible mode as disabled and chooses the next closest one. E.g. if you choose the wide lens, it doesn't support 4K 60fps so it disables the 60fps mode. Likewise if you're on 4K60 with the tele lens, it hides the wide lens selector.

…but that all seems super complex to get right, and I get annoyed by how they present it now and then.