elementary/camera #170: Feature Request : No black bars when resizing elementary camera window

eric0001, 2021-07-17:
(same feature request as https://github.com/elementary/videos/issues/231#issue-946902346)
In elementary OS 6 Beta 2 (Odin) when you open elementary camera there will be black bars on left and right side of the image canvas (or on top and on bottom). The same black bars appear when you resize the window.
Expected behavior would be : just like PiP PictureInPicture resizing behavior. No black bars appearing.

[image]
[image]

jeremypw, 2021-09-06:
Presumably you would want the window aspect ratio to be locked to that of the image rather than the image being scaled and cropped to fill the available space?

jeremypw, 2021-09-06:
I note that the Ubuntu "Cheese" app has the same issue.

jeremypw, 2021-09-07:
Attempts to adjust the geometry of the window on the fly while resizing had problems such as visual glitches but PR #184 reduces the impact of the bars by not reintroducing them every time the app is restarted if they are removed by manual window dimension adjustment.

jeremypw, 2021-09-07:
The fact that the window has a headerbar makes it more complicated than the PIP case since the required aspect-ratio of the window is not the same as that of the video and also it varies with window size.

jeremypw, 2021-09-07:
It may be possible to put the headbar in a revealer and have it overlay the video when hovered.  Then the window could be given a fixed aspect ratio that matches the video.

cassidyjames, 2021-12-06:
I believe this is a duplicate of #54… correct me if I'm wrong.

closed 2021-12-06 by cassidyjames