elementary/camera #175: Camera not working on Dell XPS 13 9370

im-barney, 2021-08-12:
Hi,

Installed eOS 6.0
The camera worked without any issues on eOS 5.1.x.
Seems not to be working now.
Installed Cheese and that works fine but it would be nice to have the default camera app working.
I just get a black "image" in Camera and the camera light doesn't come on so it seems like it isn't even recognising the hardware.

Output from lshw:
 *-usb:0
                   description: Video
                   product: Integrated_Webcam_HD
                   vendor: CN0FFMHCLOG0087TB1W8A01
                   physical id: 5
                   bus info: usb@1:5
                   version: 72.79
                   serial: 200901010001
                   capabilities: usb-2.01
                   configuration: driver=uvcvideo maxpower=500mA speed=480Mbit/s

Thanks!

jeremypw, 2021-10-29:
Do you have two cameras installed? There is a possibility of an issue related to two cameras.

casasfernando, 2021-10-29:
I have only one camera but it actually has two: [RGB + IR](https://www.dell.com/support/manuals/en-us/xps-13-9310-laptop/xps-13-9310-setup-and-specifications/camera?guid=guid-8f3aa5ee-533b-4e91-9a10-099f163cb894&lang=en-us)

lsusb output:

```
fernando@nbookfcs:~$ sudo lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0c45:6a0f Microdia Integrated_Webcam_HD
Bus 003 Device 002: ID 27c6:533c Shenzhen Goodix Technology Co.,Ltd. FingerPrint
Bus 003 Device 004: ID 8087:0026 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Camera devices (video0 is RGB, video2 is IR):

```
fernando@nbookfcs:~$ v4l2-ctl --list-devices
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-6):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
```
```
fernando@nbookfcs:~$ ll /dev/video*
crw-rw----+ 1 root video 81, 0 oct 29 21:47 /dev/video0
crw-rw----+ 1 root video 81, 1 oct 29 21:47 /dev/video1
crw-rw----+ 1 root video 81, 2 oct 29 21:47 /dev/video2
crw-rw----+ 1 root video 81, 3 oct 29 21:47 /dev/video3
```
```
fernando@nbookfcs:~$ v4l2-ctl -d /dev/video0 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : Integrated_Webcam_HD: Integrate
	Bus info         : usb-0000:00:14.0-6
	Driver version   : 5.10.68
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/720
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 1843789
	Colorspace        : Default
	Transfer Function : Default (maps to Rec. 709)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 720
	Default     : Left 0, Top 0, Width 1280, Height 720
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
[... 79 lines cut ...]
	Read buffers     : 0
fernando@nbookfcs:~$ v4l2-ctl -d /dev/video3 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : Integrated_Webcam_HD: Integrate
	Bus info         : usb-0000:00:14.0-6
	Driver version   : 5.10.68
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04a00000
		Metadata Capture
		Streaming
		Extended Pix Format
Priority: 2
Format Metadata Capture:
	Sample Format   : 'UVCH' (UVC Payload Header Metadata)
	Buffer Size     : 1024
```

Hope this helps.
Thanks.

jeremypw, 2021-10-30:
@casasfernando Thanks! If it is an integrated device it could be giving the same display name for both `video0` and `video` devices, in which case PR #190 may fix you issue.  If you are able to test that PR and leave a comment that would be great.

bmbariah, 2021-11-09:
Having the same issue with the Dell XPS 15 9500

cassidyjames, 2021-12-03:
@im-barney @casasfernando @bmbariah can you confirm whether the camera light comes on and a photo is taken when you hit the capture button? I'm seeing something similar here and wondering if it's the same issue or not.

casasfernando, 2021-12-03:
I just checked with the latest version of Camera (system up to date) and things improved a little bit.

- When I first open the app I now get to see an image (previously it was just black) and the camera light is on.
- I see to cameras listed in the app, both with exactly the same name. Only the second one in the list works (shows image and the camera light is on).

Now to answer your question @cassidyjames if I click on the capture button the image just freeze, but no picture is taken/saved and the camera light goes off (the image in the app is still there tough). The only way I found to get it back to work is to select again the second camera from the list. Then picture is live again and camera light is on.

On the other hand if I switch the app into video capture mode and I click on the capture button the camera light remains on, the image is still live -but with many random freezes- and when I click on stop, the video is saved correctly but the image freeze, the camera light remains on and I need to select again the second camera from the list to get it to work again.
If I playback the saved video, it looks exactly as while capturing it, with a lot of freezes and no audio during the freezes.

Camera version:

```
Name                                             Application ID                                       Version                 Branch                Origin                Installation
Camera                                           io.elementary.camera                                 6.0.2                   stable                appcenter             system
```

Hope this detailed description of the current status helps. :smile:

cassidyjames, 2021-12-03:
Thank you @casasfernando! It sounds like it's not the exact issue I am seeing in #198 then, so I will leave that issue open.

casasfernando, 2021-12-04:
In case it helps I just ran Camera in debug mode and this is the output when trying to capture a photo:

```
[📦 io.elementary.camera ~]$ G_MESSAGES_DEBUG=all io.elementary.camera
(io.elementary.camera:223): GVFS-DEBUG: 01:07:33.356: org.gtk.vfs.MountTracker.listMountableInfo call failed: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: org.freedesktop.DBus.Error.ServiceUnknown (g-dbus-error-quark, 2)

(io.elementary.camera:223): GLib-GIO-DEBUG: 01:07:33.356: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
(io.elementary.camera:223): GLib-GIO-DEBUG: 01:07:33.359: _g_io_module_get_default: Found default implementation keyfile (GKeyfileSettingsBackend) for ?gsettings-backend?
(io.elementary.camera:223): GLib-DEBUG: 01:07:33.359: unsetenv() is not thread-safe and should not be used after threads are created
Gtk-Message: 01:07:33.411: Failed to load module "canberra-gtk-module"
Gtk-Message: 01:07:33.412: Failed to load module "canberra-gtk-module"
(io.elementary.camera:223): GLib-DEBUG: 01:07:33.412: unsetenv() is not thread-safe and should not be used after threads are created

** (io.elementary.camera:223): WARNING **: 01:07:39.290: CameraView.vala:264: Could not make picture pipeline for photo - could not set property "brightness" in element "videobalance2" to "0,000000"
```

And below, the output when I re-select the second camera from the list to get it back to work:

```
(io.elementary.camera:223): GStreamer-CRITICAL **: 01:07:58.092: gst_element_unlink: assertion 'GST_IS_ELEMENT (dest)' failed

(io.elementary.camera:223): GStreamer-CRITICAL **: 01:07:58.093: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(io.elementary.camera:223): GStreamer-CRITICAL **: 01:07:58.093: gst_bin_remove: assertion 'GST_IS_ELEMENT (element)' failed

(io.elementary.camera:223): GStreamer-CRITICAL **: 01:07:58.093: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed

(io.elementary.camera:223): GLib-GObject-CRITICAL **: 01:07:58.093: g_object_run_dispose: assertion 'G_IS_OBJECT (object)' failed
```

Thanks.