elementary/camera #181: no element "gtksink"

davidak, 2021-08-17:
## What Happened
i get this error when i start the program on nixos

[image]

any idea how to fix it?

it is not reproducible on elementary OS (tested in live session)

## Expected Behavior
no errors

## Steps to Reproduce

1.
2.
3.

### Logs

## Platform Information

system built from this WIP PR https://github.com/NixOS/nixpkgs/pull/130380

tintou, 2021-08-17:
You need [GStreamer GTK Sink](https://gstreamer.freedesktop.org/documentation/gtk/gtksink.html?gi-language=c) at run-time I don't know which package provides it in NixOS though

closed 2021-08-17 by tintou

davidak, 2021-08-17:
@tintou thanks. that helps!

>Package – GStreamer Good Plug-ins

so it should be `gst_all_1.gst-plugins-good`

cc @bobby285271