elementary/camera #182: Camera does not work anymore

davidak, 2021-08-17:
## What Happened
since updating pantheon to version 6, the camera app does not work anymore

[image]

it worked in version 5 and still works in other programs like OBS and Chromium

## Expected Behavior
working program

## Steps to Reproduce

1. open camera app
2. see no camera picture

### Logs

```
[davidak@gaming:~]$ io.elementary.camera 

** (io.elementary.camera:46453): WARNING **: 13:13:49.522: CameraView.vala:104: CameraView: Attempt to add camera that is already in list

(io.elementary.camera:46453): GLib-GObject-WARNING **: 13:13:53.578: ../gobject/gsignal.c:2732: instance '0xfc81c0' has no handler with id '205'
```

## Platform Information

nixos built from this WIP PR https://github.com/NixOS/nixpkgs/pull/130380

also reproducible on elementary OS 6, in the live session

Camera

Bus 001 Device 002: ID 046d:08b2 Logitech, Inc. QuickCam Pro 4000

tintou, 2021-08-17:
Works well from here, have you tried with GNOME Cheese to see if it works there

davidak, 2021-08-17:
@tintou good idea!

it actually works

[image]

jeremypw, 2021-10-29:
@wout  If you are able to test it, it would be helpful for you to try out PR #190 to see whether it fixes your issue.

wout, 2021-10-30:
@jeremypw Thanks for taking the time to look at this. I'll try it out tomorrow and report back.

wout, 2021-10-31:
@jeremypw Just tested the `allow-devices-same-name` branch and it's working as expected now. Thanks for the fix!

PR #190 (2021-11-16): Do not rely on display name to distinguish devices
Fixes #182 
Fixes #186 

This PR is intended to fix some problems related to having two cameras installed (see https://github.com/elementary/camera/issues/182#issuecomment-953182535 for something this might fix).  If the cameras have the same name or if `start_view` is re-entered before the pipeline is setup then no picture is shown.  Some code clean up and simplification was also done, saving some code and reducing the number of entities.

* Pass Device objects between functions and in signals
* Lose unneeded list of devices - store in menuit [... cut]
closed 2021-11-16 by jeremypw