elementary/camera #186: Need to explicitly select camera

Waester, 2021-09-12:
## What Happened

On my laptop (HP EliteBook 830 G5) my camera won't turn on in io.elementary.camera unless I explicitly select it.

I say explicitly because in the GUI of the camera app I can already see that it is selected by default when I run the camera app (by the indicated checkmark).

## Expected Behavior

I expect the camera to turn on when it is already select by the camera app as the default camera.

## Steps to Reproduce

In my case I encounter the issue just by opening the camera app.

### Logs

N/A, the guide on https://elementary.io/docs/code/os-dev#debug-logs does not work with flatpak applications.

## Platform Information

HP EliteBook 830 G5
elementaryOS Odin
io.elementary.camera (Latest release)

meisenzahl, 2021-11-15:
I have the same issue. @Waester do you have in the dropdown menu also 2 or more cameras?

[image]

Waester, 2021-11-15:
@meisenzahl 
Yes

PR #190 (2021-11-16): Do not rely on display name to distinguish devices
Fixes #182 
Fixes #186 

This PR is intended to fix some problems related to having two cameras installed (see https://github.com/elementary/camera/issues/182#issuecomment-953182535 for something this might fix).  If the cameras have the same name or if `start_view` is re-entered before the pipeline is setup then no picture is shown.  Some code clean up and simplification was also done, saving some code and reducing the number of entities.

* Pass Device objects between functions and in signals
* Lose unneeded list of devices - store in menuit [... cut]
closed 2021-11-16 by jeremypw