elementary/camera #193: Camera files are zero byte files associated with code application. Video and stills affected.

canopusStar, 2021-11-05:
### What Happened?

Camera images and videos are saved as zero byte files (2021-11-0609:41:11.jpg) which are default associated with the 'code' application in the default webcam folder.

- HP Pavillion built in webcam both video and images are affected
- DELL Opti-plex with USB webcam only images are affected
[image]

### Steps to Reproduce

1. Applications
2. Camera
3. Click 'photo' icon in app
4. Check file in pictures>webcam
5. Associate file with 'picture viewer'
6. File zero bytes & will not open

[image]

### Expected Behavior

Images or videos should be associated with appropriate applications and open accordingly when clicked

### OS Version

6.x (Odin)

### Software Version

Latest release (I have run all updates)

### Hardware Info

DELL Optiplex 980 SSD & 8 GB - cheap and nasty USB webcam only images affected
HP Pavillion AIO MS212a standard build, built in webcam both images and videos

jeremypw, 2021-11-17:
I cannot reproduce this but it sounds similar to #135 which has been fixed in the latest release.  Do you only have a single camera in each machine?  Camera is now Flatpak'd but should have access to your xdg-pictures and xdg-videos folders.

You could try running Camera in debug mode by running in a terminal:
```
flatpak run --command=sh --devel io.elementary.camera
G_MESSAGES_DEBUG=all io.elementary.io.camera
```

and noting any error messages produced when taking a photo

You can exit from the debugging shell with `exit`

Possible related to #175 ?  Some cameras appear to be two devices.  A fix for this was merged after the latest stable release.

OlivierDuchateau, 2021-12-05:
I have same problem, under FreeBSD (we still use GStreamer 1.16.x release). I found workaround:

```
-                "image/jpeg, width=%d, height=%d ! jpegdec ! ".printf (picture_width, picture_height) +
+                "videoscale ! video/x-raw,width=%d,height=%d !".printf (picture_width, picture_height) +
```

Full patch is [here](https://codeberg.org/olivierd/freebsd-ports-elementary/src/branch/6.0/multimedia/elementary-camera/files/patch-src_Widgets_CameraView.vala)

canopusStar, 2021-12-14:
> I have same problem, under FreeBSD (we still use GStreamer 1.16.x release). I found workaround:
>
> ```
> -                "image/jpeg, width=%d, height=%d ! jpegdec ! ".printf (picture_width, [...]

Thank you however there is insufficient information regarding how to use this information after following the 'full patch' link there was no readme about the steps required to apply this.

jeremypw, 2021-12-14:
@OlivierDuchateau Would you be able to submit a PR of your fix for review and potentially merging?

OlivierDuchateau, 2021-12-19:
> @OlivierDuchateau Would you be able to submit a PR of your fix for review and potentially merging?

Done!

@canopusStar, full patch is specific to FreeBSD, only part to fix your issue is line I show you.

jeremypw, 2021-12-19:
@canopusStar Please could you confirm that the issue still occurs for you using the latest release of Camera (or the daily version if you have it) and if so are you able to build a Flatpak from the linked PR and confirm it fixes the issue for you?

canopusStar, 2021-12-19:
@jeremypw 
Please could you confirm that the issue still occurs for you using the latest release of Camera (or the daily version if you have it)
- My system is clean and fully up to date now on 6.1 and camera v6.0.3 updated 13 Dec

 - and if so are you able to build a Flatpak from the linked PR and confirm it fixes the issue for you?
 - I ran your commands and the following error occurred:
 
 flatpak run --command=sh --devel io.elementary.camera
 error: runtime/io.elementary.Sdk/x86_64/6.1 not installed
 
 G_MESSAGES_DEBUG=all io.elementary.io.camera
 io.elementary.io.camera: command not found

- Additional info: single camera appears as two camera in the drop-down box in the App using either camera, the same faulted zero-byte file is generated. There is only one USB camera attached to the (DELL Optiplex 980)

jeremypw, 2021-12-21:
The duplicate camera thing is a different issue (already reported) - I also get that but it does not cause the zero-byte problem.   To run the Flatpak in debug mode you need to install the elementary Sdk - have you done that? (`flatpak install io.elementary.Sdk`)
> io.elementary.io.camera: command not found

Did you mistype the executable name (`io.elementary.camera`)?  You can use tab completion to help avoid typos.

If you look at the PR conversation you can see a link to the pre-built flatpak that you can download and install (use Files to find the downloaded file in `~/Downloads`, open the zip file by double-clicking it, Extract the `.flatpak` file and then install that by double-clicking it to open Sideload.

You can check that the camera flatpak is installed with `flatpak list`.

PR #212 (2021-12-21): Fix jpeg pipeline
Fixes #193
closed 2021-12-21 by cassidyjames