elementary/camera #198: Preview not shown on StarBook Mk V, but photos are still taken

cassidyjames, 2021-12-03:
### What Happened?

On my StarBook Mk V, the camera app shows a black preview and the hardware camera light does not turn on. When taking a photo, the light turns on briefly and a photo is saved, but no preview is ever shown.

Tested with both the latest release as a Flatpak and then `master` built and running natively.

Could be the same root cause as #149 and #175.

Before 62cf2ac37314b8eb8654e59b3fd9e647ed844d9e there is still no preview, but no photo is taken (a 0-byte file is saved)

### Steps to Reproduce

1. Have a StarBook Mk V (good luck, though testing on other hardware is helpful)
2. Open Camera
3. See no preview
4. Take a photo
5. See the photo is saved

### Expected Behavior

The preview should show up, especially since we're able to access the camera _some how_ when actually taking the photo. If we can't access it for some reason, we should show a more helpful error than a black screen.

### OS Version

6.x (Odin)

### Software Version

Compiled from git

### Hardware Info

There is only one camera (not two, and not an IR camera or anything). I don't know what all this means but hopefully it may be helpful to someone. :smile: 

```
$ v4l2-ctl --list-devices
USB 2.0 Camera: USB Camera (usb-0000:00:14.0-4):
	/dev/video0
	/dev/video1
```

```
$ ll /dev/video*
crw-rw----+ 1 root video 81, 0 Dec  3 10:07 /dev/video0
crw-rw----+ 1 root video 81, 1 Dec  3 10:07 /dev/video1
```

```
$ v4l2-ctl -d /dev/video0 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : USB 2.0 Camera: USB Camera
	Bus info         : usb-0000:00:14.0-4
	Driver version   : 5.11.22
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/720
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 1843789
	Colorspace        : sRGB
	Transfer Function : Rec. 709
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 720
	Default     : Left 0, Top 0, Width 1280, Height 720
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=18 value=18
                       contrast 0x00980901 (int)    : min=0 max=64 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=51 value=51
                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=72 max=500 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=0 max=100 step=1 default=7 value=7
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=8 step=1 default=5 value=5
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
				1: Manual Mode
				3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=157 value=157 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
```

```
$ v4l2-ctl -d /dev/video1 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : USB 2.0 Camera: USB Camera
	Bus info         : usb-0000:00:14.0-4
	Driver version   : 5.11.22
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04a00000
		Metadata Capture
		Streaming
		Extended Pix Format
Priority: 2
Format Metadata Capture:
	Sample Format   : 'UVCH' (UVC Payload Header Metadata)
	Buffer Size     : 1024
```

cassidyjames, 2021-12-03:
Maybe not the most helpful thing, but after testing a handful of suspect commits, it looks like this is broken on this hardware since a5c08d818dac9cf6c9236a7ce102edee35e13917—which makes some sense since that completely redid how the preview is displayed. The commit before (831c858dc802443151ebaded643d3a7ef5f1356d) shows the camera preview.

cassidyjames, 2021-12-06:
Debugging output:

```
$ G_MESSAGES_DEBUG=all io.elementary.camera 
(io.elementary.camera:28319): GLib-DEBUG: 13:51:04.470: posix_spawn avoided (fd close requested) 
(io.elementary.camera:28319): GLib-GIO-DEBUG: 13:51:04.629: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ?gsettings-backend?
(io.elementary.camera:28319): dconf-DEBUG: 13:51:04.629: watch_fast: "/io/elementary/camera/settings/" (establishing: 0, active: 0)
(io.elementary.camera:28319): GLib-DEBUG: 13:51:04.629: unsetenv() is not thread-safe and should not be used after threads are created
(io.elementary.camera:28319): dconf-DEBUG: 13:51:04.630: watch_established: "/io/elementary/camera/settings/" (establishing: 1)
(io.elementary.camera:28319): GLib-GIO-DEBUG: 13:51:04.636: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
(io.elementary.camera:28319): GLib-DEBUG: 13:51:04.656: unsetenv() is not thread-safe and should not be used after threads are created
(io.elementary.camera:28319): Gtk-DEBUG: 13:51:04.656: Connecting to session manager
(io.elementary.camera:28319): dconf-DEBUG: 13:51:04.881: change_fast
(io.elementary.camera:28319): dconf-DEBUG: 13:51:04.881: change_notify: /io/elementary/camera/settings/window-maximized
(io.elementary.camera:28319): dconf-DEBUG: 13:51:04.881: change_fast
(io.elementary.camera:28319): dconf-DEBUG: 13:51:04.881: change_notify: /io/elementary/camera/settings/window-size
```

cassidyjames, 2021-12-06:
And with GStreamer debugging:

```
$ G_MESSAGES_DEBUG=all GST_DEBUG=2 io.elementary.camera 
(io.elementary.camera:28592): GLib-GIO-DEBUG: 13:53:17.439: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ?gsettings-backend?
(io.elementary.camera:28592): dconf-DEBUG: 13:53:17.439: watch_fast: "/io/elementary/camera/settings/" (establishing: 0, active: 0)
(io.elementary.camera:28592): GLib-DEBUG: 13:53:17.439: unsetenv() is not thread-safe and should not be used after threads are created
(io.elementary.camera:28592): dconf-DEBUG: 13:53:17.440: watch_established: "/io/elementary/camera/settings/" (establishing: 1)
(io.elementary.camera:28592): GLib-GIO-DEBUG: 13:53:17.446: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
(io.elementary.camera:28592): GLib-DEBUG: 13:53:17.461: unsetenv() is not thread-safe and should not be used after threads are created
(io.elementary.camera:28592): Gtk-DEBUG: 13:53:17.462: Connecting to session manager
0:00:00.196353813 28592 0x55e847bbc860 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<v4l2src> error: Internal data stream error.
0:00:00.196369436 28592 0x55e847bbc860 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<v4l2src> error: streaming stopped, reason not-negotiated (-4)
(io.elementary.camera:28592): dconf-DEBUG: 13:53:17.790: change_fast
(io.elementary.camera:28592): dconf-DEBUG: 13:53:17.790: change_notify: /io/elementary/camera/settings/window-maximized
(io.elementary.camera:28592): dconf-DEBUG: 13:53:17.790: change_fast
(io.elementary.camera:28592): dconf-DEBUG: 13:53:17.790: change_notify: /io/elementary/camera/settings/window-size
```

Appended when taking a photo:

```0:00:04.418675961 28592 0x55e847bbc800 WARN          v4l2bufferpool gstv4l2bufferpool.c:809:gst_v4l2_buffer_pool_start:<v4l2src:pool:src> Uncertain or not enough buffers, enabling copy threshold
0:00:05.436008155 28592 0x55e847bc32a0 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<v4l2src> error: Internal data stream error.
0:00:05.436036243 28592 0x55e847bc32a0 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<v4l2src> error: streaming stopped, reason not-negotiated (-4)
```

cassidyjames, 2021-12-06:
And the massive level 5 debug output (warning, 6 MB text file with like 34,000 lines): [camera-debug-5.txt](https://github.com/elementary/camera/files/7663509/camera-debug-5.txt)

stsdc, 2022-01-26:
Interestingly it works nice on https://webcamtests.com/

cassidyjames, 2022-02-07:
As an update (and thanks to the work of @tintou), the preview has been solved in `master`, however, it has introduced the regression where high-resolution cameras with low-framerate modes are selecting the highest resolution by default, giving a choppy appearance in the preview and video recording. I _believe_ is the previous issue we saw with high-resolution webcams, not actually a performance issue but just a bad chosen resolution/framerate combination. However, the resolution cap we'd previously introduced caused the preview issue here on the StarBook.

Since the StarBook-specific issue is closed, I'm going to close this issue; however, I've opened #225 with the framerate issue which #224 _sort of_ addresses and might be the best we can do with the wide variety of hardware out there.

closed 2022-02-07 by cassidyjames