elementary/camera #225: Webcams with high resolution but low framerate modes choose too-high default resolution with no way to change it

cassidyjames, 2022-02-07:
### What Happened?

Since #219 and #220 follow-up, we've fixed camera availability and performance on a number of models. However, as a result, Camera once again defaults both the preview and capture to the highest-possible resolution, which on certain hardware (like my Logitech C920) is actually a very low framerate mode seemingly only intended for still photos.

### Steps to Reproduce

1. Plug in a Logitech C920
2. Open Camera (from `master`)
3. See the default framerate

### Expected Behavior

Camera should _ideally_ default to a higher framerate; perhaps we can select something like the highest resolution with at least 24fps? And/or for hardware compatibility, we may want to consider exposing available resolutions and framerates to enable the user to override the default depending on their hardware (like #224).

### OS Version

6.x (Odin)

### Software Version

Compiled from git

### Hardware Info

StarBook Mk V with a USB Logitech C920

alkaris2, 2022-05-14:
Using Logitech C920 but I can't change the resolution or framerate? If I open the camera with V4L2 Test utility it's able to run MJPEG 30fps 1080p which is it's native max resolution that it can run, but these settings can't be carried over to this app?

Running on Manjaro, downloaded from AUR with v6.0.3-2.

should be as simple as clicking on a settings menu to being able to set the resolution and framerate to configure more options for the camera. Instead what I see is low resolution 640x480 30fps. Couldn't you just store this setting in the GSchema Config to reference the resolution and FPS of the camera device selected?

[image]

cassidyjames, 2022-05-23:
@alkaris2 you seem to be requesting the feature that would be implemented by #224