elementary/camera #246: Instant crash on launch, segmentation fault

AI-ien, 2022-10-28:
### What Happened?

App crashes on launch.

### Steps to Reproduce

Launch Camera app.

### Expected Behavior

OS: Fedora release
Kernel: 5.19.15-201.fc36.x86_64 
GNOME Shell 42.5

### OS Version

Other Linux

### Software Version

Latest release (I have run all updates)

### Log Output

```shell
Flatpak run error:
(io.elementary.camera:2): Gdk-WARNING **: 11:55:07.846: eglMakeCurrent failed
-----
Started app-flatpak-io.elementary.camera-11648.scope.
Process 11659 (io.elementary.camera) crashed in ??()

Fedora rpm run error:
(io.elementary.camera:12556): GStreamer-CRITICAL **: 12:01:20.445: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed
(io.elementary.camera:12556): GStreamer-CRITICAL **: 12:01:20.445: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (dest)' failed
(io.elementary.camera:12556): GLib-GObject-CRITICAL **: 12:01:20.445: g_object_get: assertion 'G_IS_OBJECT (object)' failed
Segmentation fault (core dumped)
-----
Started app-gnome-io.elementary.camera-12429.scope - Application launched by gnome-shell.
Process 6420 (io.elementary.camera) crashed in gtk_container_add()
```

niko2, 2023-01-28:
same issue on latest release (6.2.1-1)

OS: Manjaro Linux
Kernel: 6.1.7-1-MANJARO (64 bit)
KDE Plasma 5.26.5

4piu, 2023-02-15:
Same error here:
```
(io.elementary.camera:45747): GStreamer-CRITICAL **: 14:57:07.393: gst_bin_add: assertion 'GST_IS_ELEMENT (element)' failed

(io.elementary.camera:45747): GStreamer-CRITICAL **: 14:57:07.393: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (dest)' failed

(io.elementary.camera:45747): GLib-GObject-CRITICAL **: 14:57:07.393: g_object_get: assertion 'G_IS_OBJECT (object)' failed
[1]    45747 segmentation fault (core dumped)  io.elementary.camera
```

OS: Manjaro Linux x86_64 
Kernel: 6.1.11-1-MANJARO 
DE: Plasma 5.26.5