elementary/camera #269: camera v6.2.2 crashes on launch on Fedora

decathorpe, 2024-01-31:
I got a crash report that Camera crashes on start on Fedora 38 (GNOME/Wayland), and I was able to reproduce the same issue on Fedora 39 (both GNOME/Wayland and GNOME/Xorg) - I don't have a fully functional Pantheon session yet where I can test this, but I don't think Pantheon and GNOME/Xorg should be different here.

On Wayland, there are some Gdk-WARNING log messages, and on Xorg, there are GStreamer-CRITICAL log messages, but the crash that immediately follows those looks similar (null pointer dereference?).

I ran `valgrind io.elementary.camera` on both GNOME/Wayland and GNOME/Xorg on Fedora 39 and these are the logs that also includes the backtraces for the invalid reads / writes (there seem to be both use-after-free issues *and* null-pointer dereferences).

<details>
<summary>log: valgrind io.elementary.camera on Wayland</summary>

```
==715736== Memcheck, a memory error detector
==715736== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==715736== Using Valgrind-3.22.0 and LibVEX; rerun with -h for copyright info
==715736== Command: io.elementary.camera
==715736== 
==715736== Invalid write of size 8
==715736==    at 0x1A1DF7D8: UnknownInlinedFun (list.h:61)
==715736==    by 0x1A1DF7D8: destroy_port (gstpipewiredeviceprovider.c:489)
==715736==    by 0x1A27BB87: pw_proxy_destroy (proxy.c:230)
==715736==    by 0x1A27BDB7: pw_proxy_remove (proxy.c:254)
==715736==    by 0x1A246D08: UnknownInlinedFun (core.c:166)
==715736==    by 0x1A246D08: UnknownInlinedFun (map.h:218)
==715736==    by 0x1A246D08: UnknownInlinedFun (core.c:205)
==715736==    by 0x1A246D08: proxy_core_removed (core.c:186)
==715736==    by 0x1A27BDB7: pw_proxy_remove (proxy.c:254)
==715736==    by 0x1A247177: pw_core_disconnect (core.c:487)
==715736==    by 0x1A1DC777: gst_pipewire_core_release (gstpipewirecore.c:174)
==715736==    by 0x1A1DF3B6: gst_pipewire_device_provider_probe (gstpipewiredeviceprovider.c:624)
==715736==    by 0x48D5C8F: gst_device_provider_get_devices (gstdeviceprovider.c:424)
==715736==    by 0x48EAC79: gst_device_monitor_get_devices (gstdevicemonitor.c:447)
==715736==    by 0x1158CD: UnknownInlinedFun (CameraView.vala:162)
==715736==    by 0x1158CD: camera_main_window_constructor (MainWindow.vala:113)
==715736==    by 0x49DEA4B: g_object_new_with_custom_constructor (gobject.c:2163)
==715736==  Address 0x17f293c0 is 288 bytes inside a block of size 304 free'd
==715736==    at 0x4845B2C: free (vg_replace_malloc.c:985)
==715736==    by 0x1A246D08: UnknownInlinedFun (core.c:166)
==715736==    by 0x1A246D08: UnknownInlinedFun (map.h:218)
==715736==    by 0x1A246D08: UnknownInlinedFun (core.c:205)
==715736==    by 0x1A246D08: proxy_core_removed (core.c:186)
==715736==    by 0x1A27BDB7: pw_proxy_remove (proxy.c:254)
==715736==    by 0x1A247177: pw_core_disconnect (core.c:487)
==715736==    by 0x1A1DC777: gst_pipewire_core_release (gstpipewirecore.c:174)
==715736==    by 0x1A1DF3B6: gst_pipewire_device_provider_probe (gstpipewiredeviceprovider.c:624)
==715736==    by 0x48D5C8F: gst_device_provider_get_devices (gstdeviceprovider.c:424)
==715736==    by 0x48EAC79: gst_device_monitor_get_devices (gstdevicemonitor.c:447)
==715736==    by 0x1158CD: UnknownInlinedFun (CameraView.vala:162)
==715736==    by 0x1158CD: camera_main_window_constructor (MainWindow.vala:113)
==715736==    by 0x49DEA4B: g_object_new_with_custom_constructor (gobject.c:2163)
==715736==    by 0x49E0D29: UnknownInlinedFun (gobject.c:2243)
==715736==    by 0x49E0D29: g_object_new_valist (gobject.c:2584)
==715736==    by 0x49E155E: g_object_new (gobject.c:2057)
==715736==  Block was alloc'd at
==715736==    at 0x4849E60: calloc (vg_replace_malloc.c:1595)
==715736==    by 0x1A27B75E: pw_proxy_new (proxy.c:80)
==715736==    by 0x1A45E46D: registry_marshal_bind.lto_priv.0 (protocol-native.c:1852)
==715736==    by 0x1A1E07D7: UnknownInlinedFun (core.h:517)
==715736==    by 0x1A1E07D7: registry_event_global (gstpipewiredeviceprovider.c:510)
==715736==    by 0x1A45BBD1: registry_demarshal_global.lto_priv.0 (protocol-native.c:1826)
==715736==    by 0x1A449020: process_remote (module-protocol-native.c:1037)
==715736==    by 0x1A449807: on_remote_data (module-protocol-native.c:1071)
==715736==    by 0x1A2EEF15: loop_iterate (loop.c:496)
==715736==    by 0x1A290E16: do_loop (thread-loop.c:295)
==715736==    by 0x58F9896: start_thread (pthread_create.c:444)
==715736==    by 0x5980563: clone (clone.S:100)
==715736== 
==715736== Invalid write of size 8
==715736==    at 0x1A1DF7DF: UnknownInlinedFun (list.h:62)
==715736==    by 0x1A1DF7DF: destroy_port (gstpipewiredeviceprovider.c:489)
==715736==    by 0x1A27BB87: pw_proxy_destroy (proxy.c:230)
==715736==    by 0x1A27BDB7: pw_proxy_remove (proxy.c:254)
[... 293 lines cut ...]
==6350==    by 0x49EA052: g_signal_emit (gsignal.c:3675)
==6350==  If you believe this happened as a result of a stack
==6350==  overflow in your program's main thread (unlikely but
==6350==  possible), you can try to increase the size of the
==6350==  main thread stack using the --main-stacksize= flag.
==6350==  The main thread stack size used in this run was 8388608.
==6350== 
==6350== HEAP SUMMARY:
==6350==     in use at exit: 4,900,736 bytes in 51,336 blocks
==6350==   total heap usage: 403,787 allocs, 352,451 frees, 32,704,825 bytes allocated
==6350== 
==6350== LEAK SUMMARY:
==6350==    definitely lost: 49,176 bytes in 3 blocks
==6350==    indirectly lost: 72 bytes in 3 blocks
==6350==      possibly lost: 22,873 bytes in 288 blocks
==6350==    still reachable: 4,433,207 bytes in 48,414 blocks
==6350==         suppressed: 0 bytes in 0 blocks
==6350== Rerun with --leak-check=full to see details of leaked memory
==6350== 
==6350== Use --track-origins=yes to see where uninitialised values come from
==6350== For lists of detected and suppressed errors, rerun with: -s
==6350== ERROR SUMMARY: 17 errors from 4 contexts (suppressed: 0 from 0)
```

</details>

tintou, 2024-01-31:
Camera hard depends on "gstreamer1-plugins-good-gtk" in Fedora, this should fix the issue

decathorpe, 2024-01-31:
Nope - that package is already installed.

danirabbit, 2024-06-29:
I can confirm Camera crashes on OS 8 under Wayland as well, in Flatpak. It works fine in X11

bmr-cymru, 2025-05-19:
I believe the initial report of a crash on startup is resolved in (at least) 8.0.0 - I'm running elementary-camera-8.0.0-2.fc42.x86_64 with Wayland/Gnome and the application starts up fine and works correctly (it selects the V4L2 devices by default, which do not seem to work, but that may be a system issue. Selecting the underlying devices from the camera menu works perfectly). Video and still capture works as expected.

While trying to understand the video source issue I tried running `io.elementary.camera --gst-debug-level=9` which (after generating a huge amount of log messages) segfaults before initialising the window:

```shell
$ io.elementary.camera --gst-debug-level=9
...
0:00:00.531803432 626777 0x5584392659b0 TRACE        GST_REFCOUNTING gstobject.c:425:gst_object_finalize:<pipewiredevice7> 0x7f9474021870 finalize
0:00:00.531812152 626777 0x5584392659b0 TRACE        GST_REFCOUNTING gstminiobject.c:660:gst_mini_object_unref: 0x7f947400d790 unref 1->0
0:00:00.531818604 626777 0x5584392659b0 TRACE              structure gststructure.c:771:gst_structure_free: free structure 0x7f9474016690
0:00:00.531824939 626777 0x5584392659b0 TRACE               GST_CAPS gstcaps.c:219:_gst_caps_free: freeing caps 0x7f947400d790
Segmentation fault (core dumped)
```

Cycling through the debug levels this occurs with any `--gst-debug-level` value >= 5:

```shell
$ io.elementary.camera --gst-debug-level=5
...
9.999999999, seq-num 7, element 'pipewiredeviceprovider0', GstMessageDeviceRemoved, device=(GstDevice)"\(GstPipeWireDevice\)\ pipewiredevice6";
0:00:00.370640606 627074 0x559c5f1589b0 DEBUG                GST_BUS gstbus.c:452:gst_bus_post:<bus1> bus is flushing
0:00:00.370655877 627074 0x559c5f1589b0 DEBUG                GST_BUS gstbus.c:339:gst_bus_post:<bus1> [msg 0x7f999000b3f0] posting on bus device-removed message: 0x7f999000b3f0, time 99:99:99.999999999, seq-num 8, element 'pipewiredeviceprovider0', GstMessageDeviceRemoved, device=(GstDevice)"\(GstPipeWireDevice\)\ pipewiredevice7";
0:00:00.370661762 627074 0x559c5f1589b0 DEBUG                GST_BUS gstbus.c:452:gst_bus_post:<bus1> bus is flushing
Segmentation fault (core dumped)
```

@decathorpe pointed me at this issue (which I'd read previously), but I think this may be a different problem than the original report. It's obviously a minor issue since it only affects high GST debug levels when run from the terminal. Starting the app from the desktop menu runs without issue.

I'm unable to reproduce the segfault with other GST apps when using `--gst-debug-level=9` (totem, rhythmbox).

I'll try to test today with the current main via `flatpak-builder`.

I'm happy to file a separate issue if it's useful, and I can provide stack traces from the core dumps, run under valgrind etc.