elementary/camera #297: New features

manu20247, 2025-08-13:
### Problem

The app could have a background blur effect and another to change the background to any other, which would make things much easier for digital content creators, teachers, therapists, and the like. Background blur and a virtual background while recording would be two important filter "features," so we're not held hostage by other third-party apps.

### Proposal

Add 2 new filters.

1. Use virtual background
2. Blur background

danirabbit, 2025-08-13:
I think instead of adding these things directly to camera, they should probably happen somewhere in between your hardware camera and whatever software is consuming a camera stream so that you can use these kinds of features whether you're using FaceTime, Zoom, Jitsi, or whatever. I'm not sure what kind of facilities pipewire has for us messing with the stream in between it being consumed by apps, but it seems like that would be the better long-term way to go

manu20247, 2025-08-13:
Wouldn't openCV help? However, in that case it's better to study the tools we have well. I'll see if I can help. I'm still studying programming. I have some difficulties ^_^. Thank you for your attention.